RichCattell/Marlin #19: m119

prophetmaster, 2015-01-01:
hello, I have a problem with your firmware when I send a signal M119 z-probe returns me tjs open yet so I plug zmax it works. Is this a bug, or is it that I am doing something wrong?

RichCattell, 2015-01-01:
Which build are you using master or development ?
I am currently making some changes the development branch and so I might not work correctly at the moment..

RichCattell, 2015-01-01:
Which build are you using master or development ?
I am currently making some changes to the development branch and so It might not work correctly at the moment..
I am planning to release a new test build soon once I have the bugs fixed in development..

prophetmaster, 2015-01-01:
development

prophetmaster, 2015-01-01:
i have tested in master branch all of the endstop work :-)

closed 2015-01-01 by RichCattell