RichCattell/Marlin #23: Bed detect problem

designermx2, 2015-01-13:
Hi Rich. Very nice job with auto cal.

I have some problems when use the automatic calibration.

1 - I use a Polyethiremid circular base upper the heated aluminium with 17cm.
      1.1 The auto calibration goes test out of the bed.
       1.2 When found a "calibrated" value, it is not correct. The radius is not the right value and endstops too.

My calibration use 0.025 and 0.015 values and i was using G30 A 215 (my real value is 214 for all arms), but it´s a infinity loop.

When finish, i need remake the calibration using only a part of the tower correction and radius correction.

Thanks

WZ9V, 2015-01-13:
For 1.1 did you set your Bed Diameter?  The source says 170 but if your bed is 170 I would use a value more like 140 or less.  On my 250mm bed I use 200mm.

For 1.2 you might want to try the braddo99 branch that has averaging.  It hes with the infinite loop probing.  I think Rich is considering merging that or something similar to it into a future release.

designermx2, 2015-01-13:
It was set for 160mm. 140 will  make a correct value for calibration? May i increase the bed value after calibration?

RichCattell, 2015-01-13:
The bed_diameter value is only used by the autocalibration function to determinate the distance from the center of the bed to probe in front of, and opposite each tower.  The closer to the edge of the bed that these points are probed the more accurate the calibration will be.