RichCattell/Marlin #31: Auto level grid seems backwards when printing

MakerFront, 2015-02-16:
This is tough to explain, but I'll try.

Love the FW btw, it's awesome, just one thing I may be missing here.

After a G28 command, I bring the head down to check the level of the bed, it is closer to the bed towards the front of the machine and further off the bed in the back. After running a G29, I find that it prints too close to the bed in the back and barely touches in the front. Maybe I'm missing something here, but it seems like it's applying the auto level measurements backwards.

I have triple checked my steps per mm and they are spot on. Also triple checked all my delta variables and verified the rods are the same length. My axes are all in the proper directions. The auto level and calibration starts at the proper location in the +X and +Y quadrant. The probe offset is correct.

Any ideas?

Thanks in advance!

RichCattell, 2015-02-16:
G29 currently has issues in this version - I diddn't actually write the code for this function, it was imported from Johann's firmware, but it seems the version that i have imported it from had issues. I must admit i did think it might be applying the values reversed back-to-front.. i'll have a look and see if i can fix this, but i'm planning on importing my changes to the autocal functions into a newer version of Marlin at some stage which will hopefully fix this issue as G29 has had some updates recently...

If you try to print without doing a G29, does it print flat/level? I'm currently not using G29 at all for this reason...

Rich

MakerFront, 2015-02-16:
I suppose I could manually level the bed. when I move X and Y around .2mm above the bed there's an obvious difference between the front and back of the bed. 

I know, I know, it should be completely calibrated and flat and square. I guess I'm lazy and I like convoluted processes!

If you need a guinea pig for the reversed values, I'd be game!

If you do wind up writing your own auto level program, would it be possible to start at the center and work out in concentric circles? This would alleviate any issues where some of us have limitations on the towers (using linear rail) where the probe offset cannot actually move the effector to where it needs to go. This results in a skewed bed leveling grid towards the back.

Thanks!

RichCattell, 2015-02-16:
Have you tried running the Autocalibration to setup your printer?

G30  will display the current state of your printer
G30A  will attempt to automatically calibrate the printer Adjusting all values as necceary..
Once it has finished type M500 to save the new values.

This only needs to be done once to calibrate the printer (unless something changes).

G29 is autolevel which is different to the G30 Autocalibration function.

RichCattell, 2015-02-16:
Also.. make sure your using the "testing" branch of my firmware to get the latest updates to the Autocalibration

MakerFront, 2015-02-17:
I'll try the files in the "Testing" branch later today.

Thanks!
We all appreciate it!