RichCattell/Marlin #32: Autocalibration Error - Returning -20.0000 mm when thrusting and not activating FSR

timrhavens, 2015-03-01:
So, I recently build a delta printer, and am trying to use the FSR's to perform an autocalibration. I'm getting a bizarre behavior that I can't figure out. I'll post the output from pronterface:

Connecting...
start
Printer is now online.
echo: External Reset
Marlin 1.0.0
echo: Last Updated: Mar  1 2015 15:21:32 | Author: (K250VS-MB-FSR-BOM)
Compiled: Mar  1 2015
echo: Free Memory: 2551  PlannerBufferBytes: 1232
echo:Stored settings retrieved
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z80.00 E800.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z500.00 E200.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X4500 Y4500 Z4500 E500
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1500.00 T1000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X2.00 Z2.00 E2.50
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta Geometry adjustment:
echo:  M666 A0.00 B0.00 C0.00 E0.00 F0.00 G0.00 R171.50 D288.00 H233.00 P0.00
echo:PID settings:
echo:   M301 P18.62 I1.37 D63.14
echo:SD init fail
SENDING:G28
>>> m502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>> m500
SENDING:M500
echo:Settings Stored
>>> g28 a d288.0
SENDING:G28 A D288.0
>>> g30 a d288.0
SENDING:G30 A D288.0
Starting Auto Calibration..
Calibration precision: +/-0.040mm
|
-0.4000
-20.0000
-20.0000 		Exact Match: -20.0000	Range:  19.6000 mm
|
-9.5875 -6.6000 -3.6125 -0.6250  2.3625  5.3500  8.3375 		Mode/Median: -0.6250	Range:  17.9250 mm
|
-19.5500 -16.5625 -13.5750 -10.5875 -7.6000 -4.6125
-19.7750 		Mode/Median: -13.5750	Range:  15.1625 mm
|
-20.0000
-20.0000 		Exact Match: -20.0000	Range:  0.0000 mm
|
-20.0000
-20.0000 		Exact Match: -20.0000	Range:  0.0000 mm
|
-20.0000
-20.0000 		Exact Match: -20.0000	Range:  0.0000 mm
|
-18.8375
-18.7875
-18.8000
-18.8250
-18.8250 		Exact Match: -18.8250	Range:  0.0500 mm
|	Z-Tower			Endstop Offsets
| 	-0.6250			X:0.00 Y:0.00 Z:0.00
| -13.5750 	-18.8250		Tower Offsets
| 	-20.0000			A:0.00 b:0.00 C:0.00
| -20.0000 	-20.0000		I:0.00 J:0.00 K:0.00
| 	-20.0000			Delta Radius: 171.5000
| X-Tower	Y-Tower		Diagonal Rod: 288.0000
Using diagional rod length: 288.00mm (will not be adjusted)
The position of the endstop switches on this printer are not within limits
Adjust endstop switches so that they are within 3mm Z-height of each other
Current Endstop Positions - X: -20.00 Y: -20.00 Z: -0.62
Autocalibration aborted

  The strange behavior starts after the autocalibrate command The printhead comes down and taps the center of the printbed once, the fsr light goes from red to green, and i get the -0.4000 value. The head then moves up about 30 mm, then travels almost halfway to the z column of the printer. It then moves twice back down toward the printbed, but each time it stops short of the printbed, the FSR never triggers (LED stays green the whole time), and it returns -20.0000 twice. 
  The printhead then proceeds to various spots on the printbed, sometimes tapping the bed, activating the FSR, and returning real values, but most of the time just thrusting down toward the bed without actually contacting it. Whenever it fails to hit the actual bed, it returns -20.000. 
  At this point I'm just about out of ideas. Any help would be appreciated.
Tim

igranderojo, 2015-03-01:
timrhavens, Yes, I have been having the same problem.  I am using the nozzle as the probe.
I bought JohnSL PCB v1.2 for FSRs. I soldered the pins to the board.  Then wired the FSRs to the board.  Then wired the board to the RAMPS.  I have checked and rechecked that the FSRs trigger properly.  That my endstops are triggering properly.  I have followed the prep steps to make sure the FSRs and the Board work properly.  My Mini Kossel has 350mm horizontals, which makes the Delta radius different than a standard Mini Kossel.  I've tried different versions of the firmware, RichCattell and a vendor.
I am going to try bradd099 version next.  
Mine will fail to hit the glass and return -20.0 sometimes. instead of converging it is diverging.
My sense is that I have the geometry parameters wrong in the firmware or the M666 values wrong and it can not converge on a solution.  Even changing the sensitivity to make is "less" sensitive A=0.1, and NO jumpers on the JohnSL PCB.
very frustrating to have one Kossel with FSRs and auto cal working and one not working.

WZ9V, 2015-03-01:
Make sure your initial z-height is larger than the printers build height.  Marlin does not like to go below zero on z-height.  I think there is a parameter you can set to allow it to go past zero but I usually just set the build height to a bit more than it is.  The calibration will adjust the z-height so it's OK if it starts about 10 to 20 mm too high.  I wouldn't go more than that though.

igranderojo, 2015-03-02:
Does not matter what I do to change z height, delta radius, or use different firmware.  I have tried Ultibots, Rich Cattell, braddo99, and I have gotten the same behavior every time.
I have checked my circuits from the JohnSL pcb to the ramps, tested my FSRs. 
The only thing I can figure is that the dimensions on my Mini Kossel are  outside the range of expected values for the auto calibration. So the auto calibration will not work for just any Delta. But why?
What are the limits?
I tried starting with a larger Z height, it just crashed into the bed because the Z height was too much.
I finally did a manual calibration myself and got my endstops, z height, correct.
I have a flat print surface.  And tomorrow I am going to start printing again.
Taking the FSRs off the mini kossel.  They work on my Ultibots K250SV.
Why not on the Mini Kossel.  I will check the circuits, continuity, triggering and verify outside the Mini Kossel that the FSRs and PCB work.  Then I might try again in a year or two.
very frustrating.

igranderojo, 2015-03-02:
My Mini Kossel has 15X15 extrusions and started as a standard 240X600mm.
Then I changed the horizontals to 350mm and the diagonal rods to 340mm.
Z height is 100.0 mm. 
Why is this outside the range of the auto calibration?

timrhavens, 2015-03-06:
Nevermind, I figured out the problem. I used Arduino 1.6 to load the firmware on my device. Switching back to 1.0 has fixed it.

moonshadows72, 2015-03-16:
timrhavens, Im having the same issue with my Kossel 250 giving me -20.000 values.  Ive tried using Arduino 1.0 but ive been unable to compile the marlin firmware with it. From what I've read, Its because the size of marlin is too big with LCD suppot.  Are you using a LCD with your setup?

timrhavens, 2015-03-16:
I am using an LCD. Are you downloading both the latest firmware from his google drive, and the arduino version that is posted there? Those were the two that I used.