RichCattell/Marlin #33: Printer gets "Hiccups" when i enable display support.

LillerLars, 2015-03-10:
I don't know if this issue are on the Marlin Master 1.04 or the u8glib 1.17 (http://code.google.com/p/u8glib/wiki/u8glib)
HW: Arduino-compatible Mega2560+ RAMPS 1.4 + 12864 Discount Full Graphic Smart Controller. All Chinese import.
I have installed the u8glib according to instructions. 
Enabled the display in Configuration.h
(#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) 
Now when I print, the printer stalls maybe 0.1-0.3 sec every time the display updates - leaving ugly dots in the surface on the prints. (Both on print from PC and from SD-Card).

Any suggestions ?
Best regards
Lars.

kraczekj, 2015-03-11:
i had a similar issue, although not as frequent. it turned out that i was having a "Buffer under-run" it was a while ago so if this is wrong someone correct me,
 but i turned down my DELTA_SEGMENTS_PER_SECOND it seemed to go away
it was very interesting when i twisted the click encoder then it would hiccup on the hot end, and the screen would take a second to register that i twisted the click encoder. 
now i can quickly move through the menus and the printer doesn't stall at all.
hopefully that puts you in the right direction.

LillerLars, 2015-03-11:
Thank You for Your reply John :-) 
I will try Your suggestion, when I come home from work.... can't wait ;-)

LillerLars, 2015-03-11:
..... nearly...

kraczekj, 2015-03-11:
What happens now?

LillerLars, 2015-03-11:
The same, but only on autocalibration... I think.. Have to investigate a bit more :-)