RichCattell/Marlin #34: What could be causing large offsets for ox oy oz?

jb137, 2015-03-15:
When running the G30 A command I notice 4 of the 7 measurement points are well within range after few iterations and 3 (ox, oy and oz) are way off and don't seem to improve much:

```
c:    0.0016 
z:    0.0017
ox:  -0.0919
x:   -0.0019
oy:  -0.0769
y:    0.0091
oz:   0.1592
```

The ox, oy and oz are the positions where the opposite arm has to extend the most. With my probe on the side of the x tower oz has to extend the furthest of the three and this is also the position with the highest offset.

Could this be an inaccuracy in some parameter that is not covered by the calibration? Like motor feed rate (88.8889: GT2 / .9 degree motor  / 16 microsteps / 36 teeth), carriage has to descend the most for those positions? Or rod length, effect of length error becomes more significant when extending?

RichCattell, 2015-03-16:
If the opposite points (ox, oy, oz) are out by the same amount then this would indicate that the diagonal rod length set in the firmware is incorrect. If just one of them is out (or one more than the others) then it means that either the diagonal rods attached to that tower are a different length to the others or that tower position is out radially.  The tower radius can be adjusted manually if required by changing the I,J,K parameters with the M666 command.

RichCattell, 2015-03-23:
I've made some changes to my tower adjustment code.. could you download the latest version from the testing branch and try this again?

jb137, 2015-03-23:
Before I do that it seems to calibration sometimes converged to a solution and (depending on starting point) would diverge from a stable solution.

I will look at the code and see if I can apply the changes.

RichCattell, 2015-03-23:
The easiest way to apply the changes (if you dont want to download everything again) would be to just download the new version of marlin-main.cpp and replace this file in your Marlin folder - this is the only file that has changed.

jb137, 2015-04-05:
It works. On my first try I aborted after an hour. I changed the settings below:
```
//Speed for autocalibration travel and probing moves
#define AUTOCAL_TRAVELRATE 150 // mm/sec
#define AUTOCAL_PROBERATE 40 // mm/sec
```

And I reset all the settings using:
```
M666 X0
M666 Y0
M666 Z0
M666 A0
M666 B0
M666 C0
M666 I0
M666 J0
M666 K0
```

After this calibrating my printer worked very quick.