RichCattell/Marlin #37: target confusion

Sungod3000, 2015-04-09:
Hi,

I was installing the Pibot IR sensor for bedleveling and that works great, however I discovered 2 very odd behaviours in the process.

1. When I initially configured my new Delta I had a super convex bed shape. The difference between the middle of the bed and the point near the tower were 130mm. I uploaded the main branch marlin but could make the autolevel work there so I switched back to RCMarlin and it behaved normally.

2. Now that autolevel works here, the extrusion is completely of. If I give an extrusion command all other motors start slowly creeping upwards. All 3 XYZ at the same time at the same rate, and the extruder motor does nothing.

RichCattell, 2015-04-09:
Whenever ive seen this happen (the carriages moving up slowing when trying to extrude) it has been due to the endstops not being configured correctly in pins.h

Try typing M119 and see what values you get back for endstops and check for differences in the pins.h file for the type of electronics you are using...

Rich

UltiBots, 2015-04-09:
We have seen this when G28 has not been issued. Issue G28 to home the
carriages and try again.

Brad - www.UltiBots.com

Brad Norman
UltiBots LLC
772-257-1000
www.UltiBots.com

Sungod3000, 2015-04-10:
thanks for the answers both of you. seems to be alright now.