RichCattell/Marlin #40: "Infinite" calibration interations

cacomistle, 2015-05-19:
I used G30 A command to start auto calibration, but it seems that it's never going to stop...
I'm now running more than 30 iterations, but still not stopping. What could be the reason for that?

Steps I used:
1. M502 to load default value,
2. M500 to save value,
3. G30 A to start calibration,
4..My step 30 is never ending.. so I don't know what to do here yet.

kraczekj, 2015-05-19:
#define AUTOCALIBRATION_PRECISION 0.03 // mm you may have to make this value bigger until you get it to finish. then try running it again with a smaller precision.

RichCattell, 2015-05-20:
You can use a different precision value without changing the AUTOCALIBRATION_PRECISION in configuration.h by specifying it in the G30 A command .. e.g:

G30 A0.05

Would calibrate to within 0.05mm

Rich

cacomistle, 2015-05-21:
Thanks for all your replies! I tried to increase the precision value, and looked more closely to the command line feedback. This is what I found after some iteration:

23:52:04.508 : TowerB Radius error - adjusting

After several rounds of end stop adjusting, the calibration would move to geometry adjusting. And then it would report TowerB Radius error and then try to adjust for that. But it would go back to end stop calibration. And then it's the infinite loop.

What can I do to solve the TowerB error? Thanks!

More feedbacks here:

23:57:43.483 : TowerB Radius error - adjusting
23:57:59.073 : Z-Tower			Endstop Offsets
23:57:59.073 : 0.0200			X:-1.58 Y:-1.77 Z:-0.39
23:57:59.088 : -0.0400		0.0200		Tower Position Adjust
23:57:59.088 : -0.0100		0.0200		I:0.00 J:1.31 K:0.00
23:57:59.088 : 0.0200			Delta Radius: 103.2281
23:57:59.088 : X-Tower		Y-Tower		Diag Rod: 209.5000
23:57:59.088 : Iteration: 37
23:57:59.088 : Adjusting Endstops..

itubal, 2015-12-15:
After 6 or 7 hours, my printer ended at 99 iterations. I thing that 100  is the maximun amount of iterations. My printer was not calibrated.
What is the latest and de best branch?

applekeith, 2015-12-18:
I too am having the same issue... I increased my precision value to 0.1 and over 4 hours. in the middle of a very long Iteration 60. Gets the endstops, then I get the Tower radius error and goes back and forth... 

Any words of wisdom would be helpful.

Thanks in advance. The though of manually calibrating this delta (and I tried) is so maddening. I am hopeful for this firmware. And I do appreciate the efforts.

Keith

itubal, 2015-12-20:
Some thinks about it.
1) Delta Radius and Rod Length. Put a higher values. If you had a 180 mm rod, put 181 with M666 D181
2) Same for Delta Radius. I calculated 86 mm, put 90 .. M666 R90
3) No towers corrections. M666 I0 J0 K0

With a proximity sensor. After 27 iterations more than 3 hours ....
The calibration finished, but the results are BAD!!! The printer has the "pringles potatoes" effect.!!

applekeith, 2015-12-21:
I still have an issue, If I just run the G30 by itself, After it completes, the A tower carriage drifts down into the pulley... every time.

itubal, 2015-12-22:
After 15 hours calibrating the printer... it finished the calibration, but... with wrong parameters. I saved the parameters in EEPROM whith M500. Then I put... G30 and the bed measures was different. distint gap for each tower.

applekeith, 2015-12-24:
I ran my G30 A D230 (my arm length). It still crashed the A tower carriage in the end, but I was able capture the numbers. I manually put in the numbers in and they seem to be working. I just wish I could get the G30 command to execute properly.

itubal, 2015-12-29:
D is the "arm length"? Or the bed radius?

applekeith, 2015-12-29:
Arm length... I still do not know why it drifts in the end either...

itubal, 2016-01-15:
I doesn't work. I'n calibrating manually.

aglotrevla, 2016-10-17:
Hello everone ,
I use G30 A command to autocalibrate my delta printer. It start , after some iterations , it s writing autocalibration completed. But after that i cant do anything (sending home position or m500 or anyother,my eeprom seeting are open in marlin firmware)
what can i do , is there anyone can help me ?
Tolga

itubal, 2016-10-27:
save the settings to eeprom and restart.

Daneild, 2016-11-23:
Write down the values and reset the printer,  using M666/M500 put all in eeprom, is happening because (your computer will shutdown/standby after a while or..??) the USB power/port and the connection printer- computer will freeze.
do not know much about ysb power mannagment, gest a guess,  but try and it will work- for me it did.
Br

Whytehorse, 2017-01-20:
Mine also hit 99 iterations and then crashed(buffer overflow?) It said "waiting for user" on the lcd screen so I went through the menu and hit save into memory. After that the Z=0 X=0 Y=0 was 1.2mm too low. Yay!! I restarted with G30 A0.1 

FYI I have some effector tilting in front of the X tower and no clue how to fix it mechanically, which is why I went to automatic bed leveling. I think this is a cumulative error introduced by effector tilt and sends the code into an endless loop.

Whytehorse, 2017-02-02:
I wasn't able to resolve this bug, but I was able to do a workaround by manually calibrating using this calculator:
http://escher3d.com/pages/wizards/wizarddelta.php