RichCattell/Marlin #42: how to lock the diag_rod_length?

radzia2, 2015-05-25:
Hi Rich,

Great work on the code. So far, I have been using your latest released version and tried to update yesterday to your latest development version.

With your v1.04, you had the ability to "lock" the delta_rod_lengths by typing G30 A Dxxx, however, this doesn't seem to work anymore in the new code. Any plans on adding this, or am I using the wrong gcode to get this done?

Thanks,
R

radzia2, 2015-06-16:
it seems nobody has a clue? =)

RichCattell, 2015-06-16:
HI.. Sorry for delay in replying..
First of all, if you want to use my latest version you should use the "testing" branch - this has the latest updates.

The ability to "lock" the delta_rod_length adjustment was included in my 1.04 version mainly because there were instances where it diddn't detect this value correctly in all cases. In my newer version (the testing branch) it does detect this correctly and the ability to "lock" the value is no longer required and is therefore not included.

Hope this helps

Rich

Sberfield, 2015-12-15:
Doesn't changing the rod length affect dimensions of the parts? My rods are pretty close to 215mm but after calibration are set at 217. This, I think, causes objects to be undersized.

itubal, 2015-12-15:
My Rostock Mini has a 180 mm rods. The M30 A command (after 7 or 8 hours and 100 iterations) said D= 178.72 but It seems work better with 178.80. in both cases the objects are bigger and the holes smaller than the design.

caall99, 2016-03-07:
I too am noticing an issue regarding diagonal rod length. I measure my rods to be spot on at 215.11mm, and your auto calibration routine is setting them to some arbitrary value between 212.5.. and 214mm. All of my prints are coming out about 2-5% too large in the X/Y dimension. They almost appear "squashed". May I run the G30A auto cal routine, and then simply go into the firmware and change the diagonal rod length to the actual length after the fact, or will this screw up other things?