RichCattell/Marlin #45: how to compile

chuck787, 2015-07-15:
When I attempt to compile there are lots of missing files. I am using win7 64 and Arduino.ide 1.5 and 1.6 to install your marlin on a melzi.   Have not been  able to load firmware.  May I ask what boot loader files and which compiler any one is using.  It has been a very frustrating few months. If any one can help I would really appreciate the help and happily supply more details

RichCattell, 2015-07-15:
You need to use version 1.05 of the arduino ide to compile marlin firmware. It won't compile correctly with the later versions.

chuck787, 2015-07-15:
Thank you Rich.  It does now compile and upload with Arduino 1.0.5-r2 and installed Sanguino-0101r1.zip into hardware and #define MOTHERBOARD 66.
I now have a problem that bugged me before.  I have a panelolu2 from Think3dPrint3d, I uncommented the #define LCD_I2C_PANELOLU2 in Conguration.h.  Tried motherboard 63 and 66 and get multiple a DIO49_xxx not declared in scope, _RPORT, _PIN, etc, in lcd_init() in ultralcd.cpp.  Did locate PINL not defined in fastio but have no clue what to do.
Note: Started this with a reprappro tricolor but disconnected one board and use with single extruder.   I have two melzi ver 2.0 boards to play with, one I bought on ebay and one removed from the tricolor printer.  
 
I would appreciate any help
Chuck

chuck787, 2015-07-16:
I tracked down the compile error with  in fastio to be SDCARDDETECT set to 49 when LCD_I2C_PANELOLU2 is defined.
 
The pin.h defines SDCARDDETECT 49 of  the  Newpanel ifdef.
It appears that it should be SDCARDDETECT -1
 
I then checked Marlin.release on Github and it also does sets to -1.
Around line 1316:
 
     #endif //Panelolu2
     // chuck changed SDCARDDETECT to -1 from 49
     // #define SDCARDDETECT 49
     #define SDCARDDETECT -1
   #endif //Newpanel
#endif //Ultipanel
 
chuck