RichCattell/Marlin #47: Testing and development branches crashing into the bed at the end of the first iteration

digiexchris, 2015-10-14:
I have a Geeetech G2S, which has a bit of a funny geometry compared to a kossel. The master branch of your autocalibrate seems to work fine with it though, saving me many hours of work!

I tried calibrating to 0.04 first, but it hit the iteration limit. I was successfully able to get the master branch to calibrate my printer with G30 A0.06, but due to one tower being positioned fairly off, I still have a wave big enough to cause issues. so in order to try to get it closer, I worked on it mechanically a bit, changed the end stop adjustments to make them more repeatable, and manually reset the end stops to where they started when I first tried out the Master branch, as level and flat as I could get without tower and individual tower radius corrections. Right now, after a reset to firmware defaults, it's flat with a slight wave at half the radius of perhaps +-0.05mm aligned with the Z tower (moved in a line between the tower and the furthest part opposite the tower, and about the same aligned with the Y tower. The X tower is way off, so I end up with 0 at the center and 0.6mm off at the tower and off by the same amount and same direction opposite the tower (radius and position error of that tower, I think). then I tried the development branch to get the latest calibration algorithm with A0.05. 

I couldn't get the Z probe offset to behave in a sane manner, it seems it's moving diagonally from the z probe deploy point to the first touch point but triggering the probe before it gets there, so I set the printer's z-height much smaller than it is so that it'll at least complete that first point coming straight down. 

What happens is the first 6 points touch properly, then the 7th (at the base of the tower with the largest misalignment) drags on the bed as it approaches that location, perhaps 20mm from that location, nearly causing it to crash and skip steps. The iteration completes, the output mentions it's adjusting endstops, and then the 8th point (at the Y axis tower, I guess the first point of the next iteration) starts way high . Then the 9th point (it goes in the direction of opposite the Y tower) crashes near half the radius on the other side of center, jams, and skips a bunch of steps. The testing branch behaves the same.

Any ideas what I could be doing wrong?

mapembert, 2015-10-18:
Yes, this happened to me. Can't get past it... For those of you that understand how panicked you get when the head digs into the bed you will find this funny. The below is my trying G30, then I thought let's give G30 A a try. First round is nice... then "Oh no!" and I unplug the power!  

>>> G30
SENDING:G30
|	Z-Tower			Endstop Offsets
| 	-3.9100			X:0.00 Y:0.00 Z:0.00
| -3.2900		-3.2750		Tower Offsets
| 	-1.1200			A:0.00 b:0.00 C:0.00
| -3.7000		-3.7850		I:0.00 J:0.00 K:0.00
| 	-2.8750			Delta Radius: 131.9000
| X-Tower		Y-Tower		Diagonal Rod: 280.0000
>>> G30 A
SENDING:G30 A
Starting Auto Calibration..
Calibration precision: +/-0.100mm
|	Z-Tower			Endstop Offsets
| 	-3.9300			X:0.00 Y:0.00 Z:0.00
| -3.2900		-3.2800		Tower Offsets
| 	-1.1200			A:0.00 b:0.00 C:0.00
| -3.6900		-3.7900		I:0.00 J:0.00 K:0.00
| 	-2.8600			Delta Radius: 131.9000
| X-Tower		Y-Tower		Diagonal Rod: 280.0000
Iteration: 1
Checking/Adjusting endstop offsets
x:-3.6400 (adj:-3.6400) y:-3.8100 (adj:-3.8100) z:-4.0300 (adj:-4.0300)
X=ERROR Y=ERROR Z=ERROR
x:-24.2200 (adj:-27.8600) y:-24.2200 (adj:-28.0300) z:-24.2200 (adj:-28.2500)
X=ERROR Y=ERROR Z=ERROR
Disconnected.

mapembert, 2015-10-18:
This issue is consistent with my new delta, also.  Below is my output from Pronterface. 

Here are my steps:
 - G30
 - G30 A
 - Crashed as described above and I pulled the powercord!

>>> G30
SENDING:G30
|	Z-Tower			Endstop Offsets
| 	-3.9100			X:0.00 Y:0.00 Z:0.00
| -3.2900		-3.2750		Tower Offsets
| 	-1.1200			A:0.00 b:0.00 C:0.00
| -3.7000		-3.7850		I:0.00 J:0.00 K:0.00
| 	-2.8750			Delta Radius: 131.9000
| X-Tower		Y-Tower		Diagonal Rod: 280.0000
>>> G30 A
SENDING:G30 A
Starting Auto Calibration..
Calibration precision: +/-0.100mm
|	Z-Tower			Endstop Offsets
| 	-3.9300			X:0.00 Y:0.00 Z:0.00
| -3.2900		-3.2800		Tower Offsets
| 	-1.1200			A:0.00 b:0.00 C:0.00
| -3.6900		-3.7900		I:0.00 J:0.00 K:0.00
| 	-2.8600			Delta Radius: 131.9000
| X-Tower		Y-Tower		Diagonal Rod: 280.0000
Iteration: 1
Checking/Adjusting endstop offsets
x:-3.6400 (adj:-3.6400) y:-3.8100 (adj:-3.8100) z:-4.0300 (adj:-4.0300)
X=ERROR Y=ERROR Z=ERROR
x:-24.2200 (adj:-27.8600) y:-24.2200 (adj:-28.0300) z:-24.2200 (adj:-28.2500)
X=ERROR Y=ERROR Z=ERROR
Disconnected.

dacoach01, 2015-10-23:
Check your switch wires are not being pulled by the wire harness as the head traverses the bed.  This happened to my opto switch and resulted in frustrating code investigations until I found the open circuit caused by wire tension on the poorly supported switch leads.

MRGiacalone, 2015-10-23:
@digiexchris I don't think you are doing anything wrong. I just started using RichCattell's branch after I finally got my FSR's. Same thing happened to me, I noticed that one of the endstop offset was being set to a very high value just before the crash, just like its shown by @mapembert. This however, has only happened that one time and I have finally calibrated the print bed with branch.

Just to clarify/confirm the crash happened during the travel moves, not probing moves.

@dacoach01 After the crash I checked the wires on the FSR's as well, and decided to set them to normally closed (NC) rather than normally open (NO) that way if a wire gets disconnected the printer should not be doing proving moves.

Markts99, 2015-11-04:
Excuse me! I am using this set of code to run on my Delta Printer, but I run the G29 or G39, the Z-Probe wouldn't deploy, could you please enlight me how to enable the Z-Probe deployment. Thanks!
I set up the parameter as below but the Z-Probe wouldn't move at the start point.
// Z-Probe variables
// Start and end location values are used to deploy/retract the probe (will move from start to end and back again) 
#define Z_PROBE_OFFSET {-31, -10, -3.4, 0} // X, Y, Z, E distance between hotend nozzle and deployed bed leveling probe.
#define Z_PROBE_DEPLOY_START_LOCATION {20, 50, 30, 0} // X, Y, Z, E start location for z-probe deployment sequence
#define Z_PROBE_DEPLOY_END_LOCATION {5, 50, 30, 0} // X, Y, Z, E end location for z-probe deployment sequence
#define Z_PROBE_RETRACT_START_LOCATION {50, 60, 20, 0} // X, Y, Z, E start location for z-probe retract sequence
#define Z_PROBE_RETRACT_END_LOCATION {50, 60, 10, 0} // X, Y, Z, E end location for z-probe retract sequence

mapembert, 2015-11-04:
Mark,

Please review these links:

http://reprap.org/wiki/G-code#G30:_Single_Z-Probe

I never used G29. Have you tried G30 yet?

Mike

itubal, 2015-12-21:
Mapembert... set precision greater than 0.5. 0.1 is tooo little