RichCattell/Marlin #48: Confusion about tower names

digiexchris, 2015-10-14:
I'm a little confused about tower names. Could you clarify? My X motor is plugged into my front-left tower, my Y motor is the front-right tower, and the Z motor is the rear center tower. I tried a few manual adjustments and I think what I'm seeing is the A tower is the rear center tower, the B tower is front-left, and the C tower is front-right. 

Is that how your firmware is referring to it? Which would be the X, Y, and Z endstop in firmware?

ealott, 2015-10-19:
My Delta has that same configuration, and is experiencing the random head-crash behavior mentioned in #47 .

Any clarification on this would definitely be appreciated.

itubal, 2015-12-21:
I'm using the "testing" branch and I haven't this problem. Be sure that the motors and the endstops are connected in the right position in the board, in the same sequence.

Whytehorse, 2017-01-20:
I'm also affected by this issue. I never noticed anything amiss until I started using automatic bed leveling. I thing the readme.md should mention in the assumptions that the following image is(or isn't) assumed in the hardware wiring. https://lh4.googleusercontent.com/proxy/VW6m2nm9JcYfblNuiyvFPIofnIQpOux4bpRyKenbC0asdxnWN-DKVoUBQmM0n-ZxfpQheLC6Ygzfi1RT8oEy-ev0mUKQGg34jzJq2MCVjn4acvIQtw=w5000-h5000

I also had my z and x endstops swapped which caused head crashing.