RichCattell/Marlin #52: Auto calibration continuing despite reaching precision

dprotter, 2015-11-29:
Hey there, I've been using the auto calibration on my delta using FSRs and have had 2 issues. (I'm running everything via Pronterface)

1) when the probe repeatedly probes the edge of the bed often times the FSR directly opposite will activate when the probe lifts. I think it is related to how FSRs are tared (how the code determines a baseline resistance, then looks for changes over a certain % of baseline resistance). After repeated probing it seems like opposite FSRs are tared, so that the weight of the bed activates them. This leads to the probe repeatedly reading as "activated" and calculating a giant range.

2) if I can get the FSRs to behave (although they usually have the error above at least once an autocalibration) the autocalibraiton continues even after the correct precision appears to be reached, although looking through the code I couldnt quickly figure out how precision is defined. It works fine for looser precision, IE A0.1 or 0.08, but once I get down to around 0.05 it just keeps going.

For instance, I ran G30 D288.04 A0.05

and got this (the calibration continued until iteration 99, then gave the normal successful calibration message:

Iteration: 52
Adjusting Z-Height to: 221.34 mm..
|
-7.1250
-7.1250 		Exact Match: -7.1250	Range:  0.0000 mm
|
-7.0250
-7.0250 		Exact Match: -7.0250	Range:  0.0000 mm
|
-7.0875
-7.0750
-7.0625
-7.0625 		Exact Match: -7.0625	Range:  0.0250 mm
|
-7.0625
-7.0625 		Exact Match: -7.0625	Range:  0.0000 mm
|
-7.3750
-7.3750 		Exact Match: -7.3750	Range:  0.0000 mm
|
-7.2500
-7.2625
-7.2500
-7.2500 		Exact Match: -7.2500	Range:  0.0125 mm
|
-7.3375
-7.3375 		Exact Match: -7.3375	Range:  0.0000 mm
|	Z-Tower			Endstop Offsets
| 	-7.0250			X:-6.25 Y:-5.44 Z:-4.86
| -7.0625 	-7.3375		Tower Offsets
| 	-7.1250			A:-0.04 b:-0.21 C:0.25
| -7.0625 	-7.2500		I:0.00 J:0.00 K:-0.25
| 	-7.3750			Delta Radius: 171.8750
| X-Tower	Y-Tower		Diagonal Rod: 288.0400

then, after it reaches the 99th iteration:

Iteration: 99
Adjusting Z-Height to: 221.34 mm..
|
-7.1000
-7.0875
-7.1125
-7.1125 		Exact Match: -7.1125	Range:  0.0250 mm
|
-6.9750
-7.0000
-7.0125
-7.0250
-7.0250 		Exact Match: -7.0250	Range:  0.0500 mm
|
-7.0000
-7.0000 		Exact Match: -7.0000	Range:  0.0000 mm
|
-7.0375
-7.0375 		Exact Match: -7.0375	Range:  0.0000 mm
|
-7.3750
-7.3875
-7.3625
-7.3750
-7.3625
-7.3625 		Exact Match: -7.3625	Range:  0.0250 mm
|
-7.2625
-7.2625 		Exact Match: -7.2625	Range:  0.0000 mm
|
-7.3000
-7.3125
-7.3250
-7.3250 		Exact Match: -7.3250	Range:  0.0250 mm
|	Z-Tower			Endstop Offsets
| 	-7.0250			X:-6.25 Y:-5.44 Z:-4.86
| -7.0000 	-7.3250		Tower Offsets
| 	-7.1125			A:-0.04 b:-0.21 C:0.25
| -7.0375 	-7.2625		I:0.00 J:0.00 K:-0.25
| 	-7.3625			Delta Radius: 171.8750
| X-Tower	Y-Tower		Diagonal Rod: 288.0400
Auto Calibration Complete
Issue M500 Command to save calibration settings to EPROM (if enabled)

Thanks for the help!

Sberfield, 2015-12-15:
Are you using an endstop board? It shouldn't matter if multiple switches trigger - any switch triggered should read as a trigger and cause the probe to move on. Are you sure your bed isn't shifting up and down when probed? That could throw things off.

itubal, 2015-12-15:
I'm using a proximity sensor and i'm getting the same "error" after 99 iterations. I'm using a switches for endstops, not optical sensor.

itubal, 2015-12-21:
With the "testing" version and a proximity sensor and after 27 iterations, the program said that the printer was calibrated and... the printer was NOT calibrated. LOL