RichCattell/Marlin #55: Z-Probe Drags during Autocalibration

smokinpuppy, 2016-02-28:
Hi,

Great project, I can't wait to get it working on my 2020 Kossel. I've set everything up according to the settings described in the configuration.h file, but when my printer is going through the G30 command, it drags the Z-probe (Spring Loaded allen key) on the bed. Where can I add the z_raise_between_probing setting so that doesn't happen?

Thank you,

Robert

smokinpuppy, 2016-03-02:
Is this project still active?

itubal, 2016-03-03:
I had the same problem, I saw this parameter and I changed it but now I don't remember. Later I'll see the font program and I'll send you the parameter and the line that you can modify.

nzinov, 2016-09-18:
I solved this problem by changing Z_MIN_ENDSTOP_INVERTING constant in Configuaration.h. You can check this by issuing M119 command  - it should read "open" when probe is not touching the bed. Hope this helps!